albertaloop/T_SWE_2019_2020 #41: Prototype temperature monitors

shehraj123, 2022-02-07:
3 temperature sensors need to be set up for basic input and output - 

- [ ] TMP36
- [ ] MAX6675
- [ ] BME280

The code for I/O can be found [here](https://drive.google.com/drive/u/2/folders/1DaQy2Y4JOMzE0vqpQm3Dud2qFFLeYrqr) (see sensor testing and code).

Tasks Remaining -

- [ ] Calibrate the sensors for better accuracy
- [ ] Connect the sensors to RSPi server

mbardwell, 2022-02-08:
@shehraj123 would you like a git tutorial? Ideally, the code would be added via Pull Request.

Please structure your issues with Summary and Acceptance Criteria sections. For example

# Summary
We need temperature sensors because... They will be located... 

# Acceptance Criteria
* [ ] breadboard sensors
* [ ] quick prototype script for TMP36
* [ ] " for MAX6675
* [ ] " for BME280
* [ ] decide on best option
* [ ] write functional test for chosen sensor

mbardwell, 2022-03-15:
FYI the Teensy's are labelled. In `/dev/serial/by-id/` you will see two devices
* 10133140 is GPS
* 10135960 is temperature sensor

mbardwell, 2022-03-19:
@Kevin-was-here good work session today. Please make a pull request with the functional test code and update any related documentation as needed.

mbardwell, 2022-03-29:
@Kevin-was-here can you make another PR and add in the TMP36 code? Example [here](https://drive.google.com/file/d/1yrk8lMAxcpcNoeGU1WCYjPCY-dKPOKKF/view?usp=sharing)