albertaloop/T_SWE_2019_2020 #42: GPS functional tests

mbardwell, 2022-02-08:
# Summary
It's nice to have automated tests that check the validity of the GPS data coming from our sensor.

# Acceptance Criteria
* [ ] identify which information is critical to pod performance
* [ ] write functional tests to assert the validity of that info. Something that compares GPS data to a 3rd party source maybe*
    * if the comparison is too hard, we can brainstorm other options
* [ ] CI?

Iyury1, 2022-06-14:
Test summary documented on Google Drive: Albertaloop\1 - Technical Division\1 - Electrical Team\1 - Software Group\6 - Testing Documents\T_SWE_TESTING - GPS.gdoc

GPS will have to be replaced because this product is reliant on the Arduino Serial.Print() function, which stops working when any other interrupt source occurs.

mbardwell, 2022-06-14:
> Test summary documented on Google Drive: Albertaloop\1 - Technical Division\1 - Electrical Team\1 - Software Group\6 - Testing Documents\T_SWE_TESTING - GPS.gdoc
>
> GPS will have to be replaced [...]

Could you push your code and link to any resources you are using to debug this problem?

Iyury1, 2022-07-16:
I added some more results for stationary GPS measurement outputs to the following file on our Google drive:
Albertaloop\1 - Technical Division\1 - Electrical Team\1 - Software Group\6 - Testing Documents\Restuls\GPS-2022-07-15.gdoc

We need to copy these results into a txt files and parse the input into a Python file so we can plot the lat-lon coordinates and calculated distance.