albertaloop/T_SWE_2019_2020 #47: Radio selection and basic testing plan

mbardwell, 2022-02-22:
# Summary
We would like to purchase our radio equipment sooner than later, but before we can agree on a date to purchase any equipment, we need a few things.
1. Hardware compatability. We want to guarantee radio endpoints can communicate with each other. 
    * Are they in the same band?
    * Are they using the same protocol?
    * How are we powering the pod-side radio + antenna?
    * How are we powering the base station side radio + antenna?
1. Connection diagram - pod and base station side 
1. Basic software plan. 
    * What applications will we need on the client? On the pod?
    * What minimum hardware is required to test the radios. Do we even need antennas if the radios are in the same room?
    * Where can we test the radios? The are Canadian air wave laws that need to be adhered too

Antenna options:
* [This](https://www.pc-canada.com/item/AMO-2G10.html) is too big
* [This](https://www.amazon.ca/dp/B07FCN6WGX/ref=emc_b_5_t) may be just right, but it requires 40W so I doubt we can connect it directly to the Pi

# Acceptance Criteria
* [ ] spec wireless hardware ensuring compatibility
* [ ] connection diagram
* [ ] software applications planned out
* [ ] idea of what minimum hardware we need to test radios planned out

mbardwell, 2022-04-19:
Alex and @OkTerrific will be meeting this week to plan hardware setup and create a testing plan

mbardwell, 2022-06-14:
# 20220613 Discussion
From [here](https://drive.google.com/file/d/1GLpjWtaKL4-aZUJ4_05yCMtyR6hs2MnM/view?usp=sharing)
[image]

* which computer will act as DHCP server?
* does the WAP need a controller?

FalsettiM, 2022-07-03:
Added instructions on how to set up DHCP server and UniFi controller to the Google Doc [here](https://docs.google.com/document/d/1bJ0HEE3l8LHQcRArAt3jzohzqHX3mB-KYA4JLHJ8hjs/edit#). Also updated the connection diagram.
[image]