albertaloop/T_SWE_2019_2020 #59: Teensy Ethernet for wired laptop-to-pod setup

mbardwell, 2022-04-12:
# Summary
Our wired setup looks like this. We need a Teensy ethernet cape to connect the Teensy to a laptop via an cat5e/6 Ethernet cable.

[image]

There are existing Teensy Ethernet modules available. They look like this.
[image]

# Acceptance Criteria
* [ ] hardware selected
* [ ] hardware purchased
* [ ] hardware put together
* [ ] basic software testing

mbardwell, 2022-04-12:
Alex purchased 5 of these https://www.pjrc.com/store/ethernet_kit.html. They are for the **Teensy 4.1**.

mbardwell, 2022-04-12:
Tried this https://github.com/vjmuzik/NativeEthernet/tree/master/examples/WebServer. It loads on the Teensy just fine (ie. we see `Ethernet WebServer Example`) over serial, but the server does not respond to http requests.

To find your Ethernet controller's MAC address, follow [this](https://forum.pjrc.com/threads/62932-Teensy-4-1-MAC-Address).

Suggested code changes to the WebServer
```
byte mac[] = {
  0x04, 0xe9, 0xe5, 0x11, 0xb4, 0x75
};
IPAddress ip(129, 128, 214, 177);
```

HTTP request
```
pi@raspberrypi:~ $ curl -v http://129.128.214.177:80/
```

mbardwell, 2022-04-20:
Made some more progress today. From reboot, I can temporarily receive UDP packets over Ethernet. After some time, it stops working

```
Starting Ethernet with IP: 129.128.214.100 and MAC: 04:E9:E5:11:B4:75
Starting UDP. Listening to port: 31548
Received packet of size 13
From 129.128.214.121, port 52182
Contents:                                         
Hello, World!                                     
Received packet of size 13
From 129.128.214.121, port 43011
Contents:                                         
Hello, World!                                     
Received packet of size 13
From 129.128.214.121, port 55248
Contents:                                         
Hello, World!                                     
Received packet of size 13
From 129.128.214.121, port 50405
Contents:                                         
Hello, World! 
```

Here's a GIF of Wireshark showing the packet is being sent to the right address and port. Click on it for a better resolution. 

[image]

To run Wireshark remotely

`$ wireshark -k -i <(ssh -l pi remotepi "dumpcap -i eth0 -P -w - -f 'udp and port 31548'")`

To send UDP packets to IP/Port

`$ python3 Documents/mike/udp_send.py`

mbardwell, 2022-05-06:
> From reboot, I can temporarily receive UDP packets over Ethernet. After 30 packets, it stops working

Gil, Alex and I debugged this today. When the Teensy is connected directly via Ethernet to the RPi with no switch in the middle, we can send UDP packets for over two minutes without any problems.

I suspect the sheer number of messages being received by the Teensy's Ethernet adaptor is causing the problem. Perhaps increasing [some buffer size somewhere](https://forum.pjrc.com/threads/67556-Teensy-randomly-hangs-when-uploading-file-via-ftp-(non-deterministic)?highlight=ethernet) could quickly fix the issue.

[hello_ethernet_ack.hex](https://user-images.githubusercontent.com/11367325/167053449-a6c3ef37-4e0c-4b30-befa-1ae23e3ae8d9.png)

Iyury1, 2022-06-21:
I modified the Network communications requirements document. 
Albertaloop\1 - Technical Division\1 - Electrical Team\1 - Software Group\4 - Requirement Documents\T_SWE_UNIT_REQUIREMENTS - Network Communications

I also slightly modified the design plan document.
Albertaloop\1 - Technical Division\1 - Electrical Team\1 - Software Group\5 - Design Plans\T_SWE_PLAN - Network Communications

Whoever will be resuming Software testing, please review our previous work on the Command and Telemetry modules located here:
https://github.com/albertaloop/T_SWE_2019_2020/tree/cody-pod-telemetry-program
https://github.com/albertaloop/T_SWE_2019_2020/tree/cody-obc/Firmware/obc

To start with we need a program for a Teensy 4.1 to send and receive UDP packets between a separate Python program. Start a new Github Branch and provide code documentation into the Design Plan document.

Iyury1, 2022-06-24:
Remote development will have to be used if you are not located in Edmonton. Here is some documentation on the remote development setup, which further links to additional documentation.
Albertaloop\1 - Technical Division\1 - Electrical Team\1 - Software Group\3 - Research Documents\Remote Development on Microcontrollers

Also take a look at the previously closed issue on Remote Development [#34](https://github.com/albertaloop/T_SWE_2019_2020/issues/34)