albertaloop/T_SWE_2019_2020 #64: GUI Introduction

mbardwell, 2022-06-14:
# Summary
This issue is a placeholder for @echeng-git to learn about the existing GUI code, get familiar with PyQT, and to come up with a plan of action that eventually unblocks #63

# Acceptance Criteria
* [ ] Go through GUI docs (talk to Ian / Dinula)
* [ ] Go through PyQT tutorials
* [ ] Come up with a plan of action

Iyury1, 2022-06-15:
All current documentation on the GUI exists in the Research Documents directory on the google drive, and in the GUI folder on our master branch.

Albertaloop\1 - Technical Division\1 - Electrical Team\1 - Software Group\3 - Research Documents\T_SWE_RESEARCH_WORKING - GUI

https://github.com/albertaloop/T_SWE_2019_2020/tree/master/GUI

Iyury1, 2022-06-15:
Started new requirements and design plan document on Google drive. 

Albertaloop\1 - Technical Division\1 - Electrical Team\1 - Software Group\4 - Requirement Documents\T_SWE_UNIT_REQUIREMENTS - GUI

Albertaloop\1 - Technical Division\1 - Electrical Team\1 - Software Group\5 - Design Plans\T_SWE_PLAN - GUI

We are going to try to open the Albertaloop_UI file in QTDesigner and make a list of all widgets that were used, and then we will decide which widgets can be removed and what we still need to add.

Iyury1, 2022-06-21:
This is a good tutorial to start with:
https://www.youtube.com/watch?v=Vde5SH8e1OQ&list=PLzMcBGfZo4-lB8MZfHPLTEHO9zJDDLpYj

Iyury1, 2022-06-25:
It just occurred to me that we shouldn't be relying exclusively on YouTube tutorials as a reference for how to use PyQT. We should definitely be reading their official documentation as well. Here is a link:
https://doc.qt.io/qtforpython/contents.html

Iyury1, 2022-07-05:
There are two standard ways we can make a table. One is using TableWidget, and the other is using TableView. The difference between a widget and a view is explained here: [https://doc.qt.io/qt-5/modelview.html](https://doc.qt.io/qt-5/modelview.html)

Basically a TableWidget would store its data within the Widget object, whereas the TableView stores its data externally as a model (as part of ModelView). If we want to have more control over our table, such as preventing too much data from being displayed at once, we will probably want to use a TableView as explained [here](https://stackoverflow.com/questions/21159095/qt-widget-for-displaying-large-amount-of-data-rows). However, I think it will more efficient to us to just create TableWidgets for now.

Here is a decent tutorial on TableWidgets: [https://www.youtube.com/watch?v=xL2NdSubiNY](https://www.youtube.com/watch?v=xL2NdSubiNY)